merlokk/SmartHome #25: ESP14MES Info?

19eighties, 2017-09-25:
Hi, do you have any information on the ESP14MES module? It looks like you are making use of it here and it would be cool to know a little more about how it works and its features. Thank you.

merlokk, 2017-09-25:
it works, but because of bad hardwafe design it measures temp of ESP(
i think how to avoid this, but it needs too much... (it needs resolder esp and activate deep sleep there)

merlokk, 2017-09-25:
maybe it needs just to decolder esp and put it on 1-2cm wires near board

19eighties, 2017-09-26:
I was thinking that could be an issue if you have the ESP chip running all the time. I tried a temperature / humidity WeMos shield and the readings were way off because of the heat from the chip.

Is there a blog post or something describing how the board works? Do you know who makes it? I am curious about how to use the ESP-14 and this module seems to be one of the few that use it that are available to buy. Thank you.

merlokk, 2017-09-26:
This module is a good idea and components but it have very bad implementation(
I dont know who produces it. I have bought it from aliexpress.
IT can work with at least 60sec deepsleep( But it cant deepsleep. It needs to add wire on esp8266 chip.