merlokk/SmartHome #27: Arduino compile error

AleGalavotti, 2018-07-27:
Hello Merlokk,
please forgive my bad english and my little knowledge of software.
I have a pair of sdm220 and  some ESP8266 I wold love to use you software that seem very professional.
I downloaded all files and try to compile with arduino IDE but I receive this message ( see below), and I don't know what to to.
Thank you in advance.
Alessandro

C:\Users\p\Desktop\Domotica\SDM\ESP8266EASTRON\general.ino: In function 'void wifiSetup(bool)':

general:102: error: no matching function for call to 'WiFiManager::WiFiManager(xLogger&)'

   WiFiManager wifiManager(DEBUG_SERIAL);

                                       ^

C:\Users\p\Desktop\Domotica\SDM\ESP8266EASTRON\general.ino:102:39: note: candidates are:

In file included from sketch\general.h:3:0,

                 from C:\Users\p\Desktop\Domotica\SDM\ESP8266EASTRON\ESP8266EASTRON.ino:17

merlokk, 2018-07-27:
Hi, it because I use my version of WifiManager. You can download it from my repository. I return home in several hours and send you link

AleGalavotti, 2018-07-29:
Hi Merlokk, 👍 really, thank you for your help.
I'll try to include your improved version (BTW I see in the sceenshots in Gti and has a very nice GUI).
Bye
Alessandro

closed 2019-06-08 by merlokk