merlokk/SmartHome #28: cant verify sketch in arduino

alexswart, 2019-06-07:
Hello,

i got different errors in arduino ide 1.8.9
i have install the necesary libary's but get the following error's

Arduino: 1.8.9 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Build options changed, rebuilding all
In file included from C:\Users\henk\Documents\Arduino\libraries\lib/xmqtt.h:8:0,

                 from C:\Users\henk\Documents\Arduino\libraries\lib\bme280.h:17,

                 from C:\Users\henk\Documents\Arduino\libraries\lib\BME280.cpp:1:

C:\Users\henk\Documents\Arduino\libraries\lib/xparam.h:66:11: error: StaticJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6

   StaticJsonBuffer<JSON_OBJ_BUFFER_LEN> jsonBuffer;

           ^

In file included from C:\Users\henk\Documents\Arduino\libraries\lib/xmqtt.h:8:0,

                 from C:\Users\henk\Documents\Arduino\libraries\lib\bme280.h:17,

                 from C:\Users\henk\Documents\Arduino\libraries\lib\BME280.cpp:1:

C:\Users\henk\Documents\Arduino\libraries\lib/xparam.h:91:11: error: StaticJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6

   StaticJsonBuffer<JSON_OBJ_BUFFER_LEN> jsonBuffer;

           ^

C:\Users\henk\Documents\Arduino\libraries\lib/xparam.h:123:11: error: StaticJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6

   StaticJsonBuffer<JSON_OBJ_BUFFER_LEN> jsonBuffer;

           ^

In file included from C:\Users\henk\Documents\Arduino\libraries\lib\bme280.h:20:0,

                 from C:\Users\henk\Documents\Arduino\libraries\lib\BME280.cpp:1:

C:\Users\henk\Documents\Arduino\libraries\Adafruit_BME280_Library/Adafruit_BME280.h:29:29: fatal error: Adafruit_Sensor.h: No such file or directory

 #include <Adafruit_Sensor.h>

                             ^

compilation terminated.

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

merlokk, 2019-06-07:
It looks that ArduinoJson released new version that incompatible with old sources. Now you can downgrade ArduinoJson and try. I'll look it when I have time(

alexswart, 2019-06-08:
I have downgrade arduinjson to 5 that errors are gone but get a lot of errors with missing things i thing i dont have the right libaries do you know what versions you are using? Or can you upload your libaries folder?

merlokk, 2019-06-08:
https://github.com/merlokk/SmartHome/releases/download/AZ7798_v_1.0/libraries.zip

alexswart, 2019-06-08:
thank you for your libaries, i have still some compile errors
what version of arduino ide are you using en wich version of esp8266 in boardmanager?

merlokk, 2019-06-08:
Arduino last. 1.8.9
esp - like 2.4.2 i have tried to check, but it clears because now https://arduino.esp8266.com/stable/package_esp8266com_index.json not in the network

alexswart, 2019-06-08:
thanx it is working :)
is it also possible to change the settings after connecting te esp to my wirless network?
i have it connected but have connect it to the wrong mqtt server can i change it?
when i go in the browser to the ip adress of the esp i dont get a website.

merlokk, 2019-06-08:
needs to press config button and it go to configure mode again
the firmware programming pin on esp

alexswart, 2019-06-09:
thank you that is working,.
i have only a small problem 1 have connect 2 sdm120 modbus kwh meters with on the last one a 120 ohm resistor, but both meters have the same id and are are baud 2400. do you know how ik can change it?

merlokk, 2019-06-09:
connect one and change its address. it have setting in the modbus address space

alexswart, 2019-06-09:
can i change it trough mqtt?

merlokk, 2019-06-10:
no, it needs to physically connect to sdm

merlokk, 2019-06-10:
40021 Network Node
http://www.eastroneurope.com/media/_system/tech_specs/3914/SDM120%20PROTOCOL.pdf
page 3
but...   there are may models of sdm and some old models have another addressing.

P.S. there is free tool QModBus for this

alexswart, 2019-06-20:
Thanks for your reply's
i have change de baud to 9600. but i dont get anny data in my mqtt from the meter
only Uptime. RSSi, VCC, Lastseendatetime and Lastbootdatetime
do i do somthing wrong? or is my ttl tu rs485 board maybe broken?

merlokk, 2019-06-21:
It looks meter don't respond. Needs to connect it to PC via rs232-ttl to USB convertor and check. Some meters have different addresses or like that

merlokk, 2019-06-21:
Convertor instead of esp

alexswart, 2019-07-25:
i have receive another rs485 converter but still no response.
on what baudrate should i set the sdm120?

merlokk, 2019-07-25:
9600 by default. there is a baud rate settings that you can change via keys on sdm.
before that you need to check polarity of rs-485.

merlokk, 2019-07-25:
connect sdm to pc and make it work. then check via esp

luke200611, 2019-07-28:
Hello,

i got  errors in arduino ide 1.8.9

error:

exit status 1
'generalSetup' was not declared in this scope. 

please help