merlokk/SmartHome #30: Cant recieve anything

thenewardu, 2020-03-23:
Hi Merlokk,

ived compiled the ESP8266EASTRON.ino and connectet as followed.
on D5 and D6 an LED
on D3 an switch

I used an RS485 Module with DE / RE Pin
"A" from Modul to "A" from SDM630
"B" from Modul to "B" from SDM630
"DI" from Modul to "TX" of ESP8266
"RO" from Modul to "RX" of ESP8266
"DE/RE" to 3.3V as Enable

When i Start the ESP866 both LED´s are on
Connected to WLAN and MQTT via IObroker

on IO Broker there is an MQTT Folder 6974CA
In That folders there some Entrys

Connected | 6974CA/Energiemeter/Connected | state | variable | OFF
and so on
The Last boot date, last seem date and connected entry is refreshing every second.

Do u know what i have to do, that my SDM630 talks to the ESP8266 ?
Do i have to modify some files ?

Thank you for your support and sry for my bad english

merlokk, 2020-03-23:
Hi

"DE/RE" to 3.3V as Enable is not good. you need the module with auto transmit control. if not - i see that you can add this functionality to softwareSerial. but its not so easy(

sometimes SDM630 (usually version 1) have different modbus register addresses (they start from 0). needs to check too...

thenewardu, 2020-03-26:
Thank you for your answer.
today arrived my module with auto transmit control but its still not working :(

I have an SDM630 V2
do i think right that rx and tx from the modul have connected to the esp rx and tx pins (GPIO1 and GPIO3 ?) 
The rx and tx LED´s of the RS485 module are not blinking or something

Do i have modify some files ? i have just compile your Eastron example without modify something.

on iobroker there are following informations, but no registers visible.

I would like to use your Eastron project so much.
Thank you
[image]

merlokk, 2020-03-26:
maybe something changes...
try this bin file https://github.com/merlokk/SmartHome/releases/tag/release_1.00

TX (IO1) needs to go to RX 485 module
RX(IO3) go to TX RS485 module
to not to break esp with wrong connection i connect them via 50R resistors

P.S. try to connect leds:
LED1 - GPIO12 ESP8266
LED2 - GPIO14 ESP8266
maybe they shows something

thenewardu, 2020-03-28:
I have uploaded your bin file on my nodemcu (NodeMCU Lua Amica Modul V2 ESP8266 ESP-12F)

both LED´s are on, but i think there is no polling, because the LED´s on the RS485 module are not blinking (there is no communication to the SDM630)

Is my nodemcu eventually not compatibel ?
Do i need an WeMos D1 Mini Pro 16MB ?

Thank you

merlokk, 2020-04-03:
usually NodeMCU cant work this way. because software uses serial port, that is already used in USB/TTL converter.
you can check if rs485 works by open serial port (usb) at speed 9600

merlokk, 2020-04-03:
I have looked at source. maybe if create SoftwareSerial here
https://github.com/merlokk/SmartHome/blob/master/ESP8266EASTRON/ESP8266EASTRON.ino#L74
and link it to eastron.SetSerial - it will work.