merlokk/SmartHome #31: error compiling

jruibarroso, 2021-01-23:
Hi, 
got this error on compiling:

Arduino: 1.8.13 (Mac OS X), Placa:"WeMos D1 R1, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

In file included from /Users/jruibarroso/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/sdk/libc/xtensa-lx106-elf/include/sys/stdio.h:6:0,
                 from /Users/jruibarroso/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/sdk/libc/xtensa-lx106-elf/include/stdio.h:63,
                 from /Users/jruibarroso/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/Arduino.h:32,
                 from /Users/jruibarroso/Documents/Arduino/libraries/autotimezone/autotimezone.h:12,
                 from /Users/jruibarroso/Documents/Arduino/SmartHome-master/ESP8266EASTRON/ESP8266EASTRON.ino:1:
/Users/jruibarroso/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:25:130: error: 'const char HTTP_HEAD []' redeclared as different kind of symbol
   #define PROGMEM __attribute__((section( "\".irom.text." __FILE__ "." __STRINGIZE(__LINE__) "."  __STRINGIZE(__COUNTER__) "\"")))
                                                                                                                                  ^
/Users/jruibarroso/Documents/Arduino/libraries/WiFiManager/WiFiManager.h:35:24: note: in expansion of macro 'PROGMEM'
 const char HTTP_HEAD[] PROGMEM            = "<!DOCTYPE html><html lang=\"en\"><head><meta name=\"viewport\" content=\"width=device-width, initial-scale=1, user-scalable=no\"/><title>{v}</title>";
                        ^
In file included from /Users/jruibarroso/Documents/Arduino/libraries/WiFiManager/WiFiManager.h:17:0,
                 from sketch/general.h:3,
                 from /Users/jruibarroso/Documents/Arduino/SmartHome-master/ESP8266EASTRON/ESP8266EASTRON.ino:27:
/Users/jruibarroso/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266WebServer/src/ESP8266WebServer.h:34:39: error: previous declaration of 'HTTPMethod HTTP_HEAD'
 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_HEAD, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };
                                       ^
/Users/jruibarroso/Documents/Arduino/SmartHome-master/ESP8266EASTRON/general.ino: In function 'void wifiSetup(bool)':
general:102:15: error: no matching function for call to 'WiFiManager::WiFiManager()'
   WiFiManager wifiManager;
               ^
/Users/jruibarroso/Documents/Arduino/SmartHome-master/ESP8266EASTRON/general.ino:102:15: note: candidates are:
In file included from sketch/general.h:3:0,
                 from /Users/jruibarroso/Documents/Arduino/SmartHome-master/ESP8266EASTRON/ESP8266EASTRON.ino:27:
/Users/jruibarroso/Documents/Arduino/libraries/WiFiManager/WiFiManager.h:77:5: note: WiFiManager::WiFiManager(TPrint&)
     WiFiManager(TPrint &prn);
     ^
/Users/jruibarroso/Documents/Arduino/libraries/WiFiManager/WiFiManager.h:77:5: note:   candidate expects 1 argument, 0 provided
/Users/jruibarroso/Documents/Arduino/libraries/WiFiManager/WiFiManager.h:74:7: note: WiFiManager::WiFiManager(WiFiManager&&)
 class WiFiManager
       ^
/Users/jruibarroso/Documents/Arduino/libraries/WiFiManager/WiFiManager.h:74:7: note:   candidate expects 1 argument, 0 provided

exit status 1
no matching function for call to 'WiFiManager::WiFiManager()'

Can you help me please ?

merlokk, 2021-01-23:
Hi
I think it too much outdated(
here:
https://github.com/merlokk/SmartHome/releases
old versions of sources and libraries
needs to compile with them or update the project to use the new libs versions

jruibarroso, 2021-01-23:
Still, can’t get it to work, installed your libs, downgraded ESP8266 library to 2.0.0 but same "no matching function for call to 'WiFiManager::WiFiManager()’ “ error

Is there any chance to you help me with this, I have a solar system with an energy meter that I need to “listen”

Thank you

merlokk, 2021-01-24:
It looks that WifiManager have not installed